samdenty/Wi-PWN #1: Version 1.3.2

br0k3nilluzion, 2017-06-16:
After Starting the Deauth attack on a network the board becomes unreachable.
The bar where is says Scan, Uers, Attack, Settings and such, becomes red, as do the buttons, and then 192.168.4.1 times out and nothing can be done until I reconnect to the WiPwn.

br0k3nilluzion, 2017-06-16:
at first I thought it was because I had the router at long range and maybe it was loosing signal.. and it seems to be holding true. 
Moved the Router closer and it doesnt happen.
So Maybe if the router signal is lost because of range or some such, the setup freezes and cant recover, even if the router comes into range again.. I'll test it more tomorrow

samdenty99, 2017-06-16:
Just to confirm:

- Once you start the Deauth attack, it disconnects from the Wi-Fi network — but are you are able to _reconnect_ and refresh the page?

When you start a Deauth attack, the module has to change the channel to that of the selected Wi-Fi network, which disconnects you from the Wi-Fi for a few seconds. Once you reconnect to the Wi-Fi, the page _should_ auto-reload and go back to normal after a few seconds.

Try the following:

1. Perform a Deauth attack and wait for your device to disconnect from the Wi-PWN network
2. Is the attack successfully working — visible LED? Or has the board became unresponsive?
3. Wait 30 seconds and then reconnect to the Wi-PWN network in your devices settings
4. Go back into your browser and wait for another 30 seconds. Does the page reload?
5. If the page doesn't reload, _try manually reloading it_

br0k3nilluzion, 2017-06-16:
Im pretty sure it was a range issue.  
The page becomes unresponsive, the title bar and buttons stay red, and taping any of the buttons does nothing.. 
Tried reconnecting but if it is still trying to connect to that Network it is still red.
I tried walking outside of the range all together, but that keeps it red and unresponsive.

To make sure it was a range issue, as far as I can tell, I tried it at a somewhat closer range and even right next to it.. it works like it should.
Then I walked out of range, more then -88 dbi or so, and it happened again.. totally unresponsive.

samdenty99, 2017-06-16:
When the page goes 'red', you should be able to refresh/reload it?

br0k3nilluzion, 2017-06-16:
Ill try it here in a couple hours again. didnt bring it to work with me. I'll try and take a video of it doing it

samdenty99, 2017-06-16:
What board are you using?

samdenty99, 2017-06-18:
Any update? I'm going to close this issue in 24 hours as I cannot replicate it.

br0k3nilluzion, 2017-06-19:
Sorry, was out of town for the weekend.
I will try and make a video today of it happening for you.

samdenty99, 2017-06-23:
It's been 96 hours, I'm closing this issue as I can't replicate it. If you are able to record a video clearly highlighting this bug, I'll re-open it.

Included in [Version 1.4](https://github.com/samdenty99/Wi-PWN/releases/tag/1.4) is a bug fix for either this problem (or one similar)

>-Bug fixes
>  -Auto reload page bug

closed 2017-06-23 by samdenty99