samdenty/Wi-PWN #2: the apk app crashes instandly

joerie11, 2017-06-20:
the app crashes on startup om my s7 edge

samdenty99, 2017-06-20:
I'm aware of this issue, currently the app only runs on API version 24 (Nougat).

The app is optional and Wi-PWN can be used exactly the same without it. Open `http://192.168.4.1` in your browser as an alternative until the app is reprogrammed to work on older version of Android

joerie11, 2017-06-20:
but as far as i know i have the newest android 7.0 nougat

samdenty99, 2017-06-20:
Can you confirm exactly what version you're running? `Settings > About > Android Version`

joerie11, 2017-06-20:
already told you Android-versie 7.0

samdenty99, 2017-06-20:
The app is currently being redesigned to have support for legacy devices. This issue should be fixed in an upcoming release (either `1.3.3` or `1.4`). I've [updated the Releases page](https://github.com/samdenty99/Wi-PWN/releases/tag/1.3.2) to highlight this bug.

joerie11, 2017-06-20:
oke

closed 2017-06-20 by joerie11

samdenty99, 2017-06-20:
I've re-compiled the APK with support for more devices.

[**RELEASE 1.3.2B**](https://github.com/samdenty99/Wi-PWN/releases/tag/1.3.2)
[Download patched APK](https://github.com/samdenty99/Wi-PWN/releases/download/1.3.2/xOptional_Wi-PWN.apk)

joerie11, 2017-06-21:
it works fine only i seem to miss the reboot button

samdenty99, 2017-06-21:
I'll make sure to re-implement the button in the next update

joerie11, 2017-06-21:
thx, You do a great job developing this