samdenty/Wi-PWN #3: Clone doesnt do anything.

xXHardGriefXx, 2017-06-21:
As the title says. When I press clone, nothing happens.

samdenty99, 2017-06-21:
When you click the clone button, it adds the network name to the below input field where you can then click on the add button

[image]

xXHardGriefXx, 2017-06-21:
It doesnt add it to the SSID textbox.

samdenty99, 2017-06-21:
What device are you using? Are you able to manually add networks with the SSID textbox?

xXHardGriefXx, 2017-06-21:
I'm using the nodemcu esp8266 and yes i am able to manually add networks.

samdenty99, 2017-06-21:
Okay, before I open a case about this can you confirm that clicking the Clone button has no effect and the input box remains the same whether you clicked the button or not?

If yes then what Operating System are you using to visit the webpage? And which browser?

xXHardGriefXx, 2017-06-21:
Do you have teamviewer? I can show you everything haha
ID: 781 224 582
password: 9t3v9c

Awnser to your quesiton: 
The button does nothing
Im using my samsung s7 edge and the browser chrome
On my computer i have the same effect. Windows 10 and chrome

samdenty99, 2017-06-21:
I have Teamviewer installed yes, does the online version of Wi-PWN work — https://wi-pwn.github.io/attack.html

xXHardGriefXx, 2017-06-21:
Yes it does. And try to connect to me via the ID and password ive posted in the message before.

samdenty99, 2017-06-21:
Issue: Special characters in network names

Patch will be released in the coming days

samdenty99, 2017-06-23:
Just released [Version 1.4](https://github.com/samdenty99/Wi-PWN/releases/tag/1.4):
><b>..</b>
>- Bug fixes
>  - ..
>  - Single quote clone bug - Issue #3
>
><b>..</b>

I've tested it with networks that have single and double quotes in the SSIDs, let me know if the patch worked :p

closed 2017-06-23 by samdenty99

xXHardGriefXx, 2017-06-23:
Working flawlessly :)