samdenty/Wi-PWN #5: buttons/shortcuts on top

thrornnc, 2017-07-10:
In the web interface and app, the buttons for scan, users, attack, settings isn't  showing (see screenshots).
[image]
[image]

samdenty99, 2017-07-10:
This may be a bug relating to the Wi-PWN minimal mode. Resetting your browsers cookie should temporarily fix this. 

Minimal mode is activated by visiting `http://192.168.4.1/?minimal=true`, did you visit this link in your browser? If not, how did you get to the Wi-PWN Web UI? By directly entering the URL?

thrornnc, 2017-07-10:
Yes, direct in url, its back! thanks.

samdenty99, 2017-07-10:
Glad you got it fixed, if you are facing any other problems just open another issue :)

closed 2017-07-10 by samdenty99