samdenty/Wi-PWN #6: Custom SSID Beacon Not Working?

ali6836, 2017-07-16:
When I click the add button to add a ssid, it doesnt add a ssid?

samdenty99, 2017-07-16:
Hey Ali, what version of Wi-PWN are you using? Can you replicate this on the [Online Interface](https://wi-pwn.github.io/attack.html)?

ali6836, 2017-07-16:
@samdenty99 I am using version 1.5 and the same thing happens on the online interface. Below, I have attached an image that shows an error that I have noticed in the code.
[image]
[image]

samdenty99, 2017-07-16:
Can you send dev-tool screenshots of the Web Interface? Haven't tested the software on any Apple products yet, but can you re-produce the steps that I would need to take to replicate this issue?

samdenty99, 2017-07-16:
Found the issue, will resolve it now :)

ali6836, 2017-07-16:
Great! How long will it take for the new release?

samdenty99, 2017-07-16:
I'll update the current release binary in about half an hour

ali6836, 2017-07-16:
Fantastic! Thanks for the help.

closed 2017-07-16 by ali6836

samdenty99, 2017-07-16:
Version 1.6 will be out in a 3-7 days, the updated binary is available to [download](https://github.com/samdenty99/Wi-PWN/raw/testing/arduino/Wi-PWN/Wi-PWN.ino.nodemcu.bin) from the testing branch

samdenty99, 2017-07-29:
[Version 6.0](https://github.com/samdenty99/Wi-PWN/releases/tag/6.0) released

> **...**
>- Fixed SSID beacon bug preventing SSIDs from being added
>- Major UI update ~
>   - Setup page redesigned, now features a cleaner design
>   - Cards have been redesigned, now feature actions at the bottom
>
> **...**