samdenty/Wi-PWN #10: Why is this thing trying to call out constantly

wcat13, 2017-08-02:
Why is afwall+ blocking this app from calling out on port 80 so much?

Curious enough to ask but to lazy to figure it out on my own.

samdenty99, 2017-08-02:
Hey @wcat13, what [version of Wi-PWN](https://github.com/samdenty99/Wi-PWN/releases) are you using? on which android version? And are you able to send screenshots

wcat13, 2017-08-02:
wow that was fast.  I don't have anything collected I just noticed it yesterday.  I installed it from play store last week.  I will try and find the time to get some screen shots for you.  It was just something I noticed because I have afwall notifying on all blocked attempts.  I glanced at it quick.  It was port 80 and it was only while it was open.

wcat13, 2017-08-02:
Ah nm I should have looked closer at it.  It is trying to hit 192.168.4.1:80, so on my lan.  I thought it strange only because I gave it permission to access my lan but not internet or data.  Sorry I should have looked at it closer.

Not sure why afwall thinks it is trying to get out on blocked route.

closed 2017-08-02 by wcat13

samdenty99, 2017-08-02:
No problem