samdenty/Wi-PWN #20: E22..

narcomerk19, 2017-09-17:
The error E22 is always showing after I start Attacking..

samdenty99, 2017-09-18:
Which version of Wi-PWN are you using?

narcomerk19, 2017-09-18:
Im using version 9.0 of your Wi-PWN.
and the boot verion was v.31_m.1, sdk v.1.3.0

narcomerk19, 2017-09-19:
@samdenty99
 Version 9.0 is the latest version right sir ?

samdenty99, 2017-09-20:
@narcomerk19 Version 9.0 is the latest preview version yes

*I'm not actively developing Wi-PWN at the moment, I may do a bit in a few weeks (it depends though)*

StijnSimons, 2017-09-24:
@narcomerk19 are you sure your antenna isn't damage and you're not far from the NodeMCU.
If this won't resolve the issue, join the [discord](https://discordapp.com/invite/yN2x7sp) for a teamviewer session / debug session.

StijnSimons, 2017-09-25:
Closing this for now. If this didn't solve your problem, reopen the issue.

narcomerk19, 2017-09-26:
I only encounter this issue in this build, I like your work coz of the material inspired 🙂
hoping to fix that bug.. my node was fully working and does not have any damage..

samdenty99, 2017-09-26:
@narcomerk19 In order to fully understand what is happening, could you please provide: 

What [version of Wi-PWN](https://github.com/samdenty99/Wi-PWN/releases) are you using? on which operating system & browser? And what are the exact steps you are doing when you face this issue? Does the attack start (LED turn on), and are you able to reconnect after?

Also, are you using the WiFi client feature? or the hotspot mode. This is a limitation of the WiFi client feature if you are

It would also be helpful if you could provide the **console output** <kbd>CTRL-SHIFT-I</kbd> → <kbd>Console</kbd>

samdenty99, 2017-10-06:
Try cloning the repo and flashing with Arduino, the binary in the `arduino` folder is of an older version, and is causing this issue

closed 2017-10-06 by samdenty99