samdenty/Wi-PWN #21: Battery charge indication

VBIralo, 2017-09-19:
Can you add a battery charge indication to the Control Panel?

br0k3nilluzion, 2017-09-20:
^^ that is a really good idea

VBIralo, 2017-09-20:
I can give a little code))

samdenty99, 2017-09-20:
I'm not actively developing Wi-PWN at the moment, I may continue in a few weeks (it depends though)

StijnSimons, 2017-09-24:
@VBIralo If you want to help out, you can send a pull request! 😉

VBIralo, 2017-09-24:
@StijnSimons I would, but I'm not very good at this. Good luck with the project! I appreciate your work! Thank you!

VBIralo, 2017-11-12:
Here, on line 148, battery voltage measurement is used -> https://goo.gl/UpTYxy

ExploiTR, 2018-06-24:
`ADC_MODE(ADC_VCC);` on top of everything in xxxxx_deauther.ino

`ESP.getVcc()` anywhere

http://esp8266.github.io/Arduino/versions/2.0.0-rc2/doc/libraries.html

Eric00x, 2019-01-24:
But there's a big problem with voltages since A0 port can support max 3.3V (in older esps it's even like 1V) and batteries (li-pos or other kinds) usually have more than 3.3V on output when fully charged.
You'll have to put voltage divider before connecting battery's power to ESP or else it will cook by itself

StijnSimons, 2019-01-25:
Closing because of compatibility issues.

closed 2019-01-25 by StijnSimons