samdenty/Wi-PWN #31: Mobile interface issue (360x640 px)

PappleTec, 2017-11-02:
On the Galaxy S7 there is an Issue with the "Bubbles" showing the signal strength, they are pointing to the false direction on this certain screen size.
<360 =  http://prntscr.com/h5dzc7 (OK)
=360 =  http://prntscr.com/h5e1a2 (ISSUE)
>360 =  http://prntscr.com/h5e1nl (OK)

StijnSimons, 2017-11-02:
Has something to do in the CSS, quite sure Sam will fix this within a day

samdenty99, 2017-11-02:
Ah yes, I'm aware of this. I had previously [asked a question on Stackoverflow](https://stackoverflow.com/questions/45355941/css-style-td-elements-depending-on-td-width) on getting the exact size when the table cells break into seperate lines, but I had use a quick JS hack (as CSS media queries can't detect the size of a DIV).

I'll try and fix this, but as there's no "fixed breaking point" it may be difficult

StijnSimons, 2017-11-02:
Also, a resolution below 300 (i think?) breaks the reboot button.

closed 2018-01-25 by samdenty99