samdenty/Wi-PWN #37: Generated virtual hot spots do not support Chinese

favgo, 2017-11-20:
Generated virtual hot spots do not support Chinese

willem640, 2017-11-24:
This is because for the SSID (hotspot name) a char is used, a char only supports ASCII characters, no chinese, actually, only characters you would use to write English, or to program in English. To change this, a String would have to be used, this introduces possible difficulties when writing it to the EEPROM (small volatile storage space in the microcontroller): in the EEPROM, you can only write one byte at a time, the String is larger than that and it s a class, which requires hacky solutions.

Currently, I do not have access to a proper computer, but in the future I may create a pull request. Or maybe not, I will see.

If you are interested in programming it yourself, you should have a look at Halleys write and read anything to EEPROM solution (http://playground.arduino.cc/Code/EEPROMWriteAnything) and lines 68 and 172 of Settings.cpp (master/arduino/Wi-PWN/Settings.cpp)

willem640, 2017-11-24:
Also, you are going to have a hard time trying to write a class to EEPROM,

This may actually be the biggest concern

edo-2313, 2017-11-24:
If it's only ASCII, why can I put emojis from my keyboard in the SSID name? I think that is Unicode, not ASCII

willem640, 2017-11-26:
Yes, I think you are right, but unicode should support chinese, so maybe look for a type (library?) That explicitly states it supports chinese

vuong244, 2018-02-25:
https://github.com/spacehuhn/esp8266_deauther/pull/644

closed 2018-03-25 by samdenty99