samdenty/Wi-PWN #38: Wifi connection to slave AP for remote administration

drdeft, 2017-12-06:
It could be very good to teach device to connect to some access point using it's WPA2 password in DHCP-client mode. Than I can forward Wi-PWN 80 port to Internet and have remote access to it.

samdenty99, 2017-12-06:
That is possible using the [WiFi client](https://wi-pwn.samdd.me/settings) feature, though it doesn't support multiple access points.

[image]

drdeft, 2017-12-06:
Wow! That's perfect project!
Will remote connection via internet work when device is in attack mode?

samdenty99, 2017-12-06:
Unfortunately the esp can only sustain simultaneous network connections whilst in standby mode, if you wanted full remote access you'd need something like a Raspberry Pi with dual network cards

drdeft, 2017-12-06:
I'm looking for Orange Pi to use it for automated WPS hacks. But esp8266 deauther is the first device I want to test. Unfortunately it has no built is RJ45 port with PoE support.

closed 2017-12-11 by samdenty99