samdenty/Wi-PWN #42: Did a translating to Polish :)

the4anoni, 2017-12-27:
Done :)

[Polish.zip](https://github.com/samdenty99/Wi-PWN/files/1588876/Polish.zip)

closed 2017-12-27 by samdenty99

samdenty99, 2017-12-27:
Translation added! You can see it live over at [wi-pwn.samdd.me/polish](https://wi-pwn.samdd.me/polish/)

the4anoni, 2017-12-27:
Oops... I fount a typo 
Corrected version
[Polish v2.zip](https://github.com/samdenty99/Wi-PWN/files/1589551/Polish.v2.zip)
Thanks for patience

samdenty99, 2017-12-27:
Fixed in [`ae95a4a`](https://github.com/samdenty99/Wi-PWN/commit/ae95a4a205f0b77e1575674108d1fc61d8e108be)

the4anoni, 2017-12-27:
But thats not all changes.
Re download file ,and check again

samdenty99, 2017-12-27:
That is the only change though

the4anoni, 2017-12-27:
No.
[image]
And one more. 
Re download file

samdenty99, 2017-12-27:
I directly downloaded and committed it, check for yourself

the4anoni, 2017-12-27:
My mistake. Again sorry.
[Polish v3.zip](https://github.com/samdenty99/Wi-PWN/files/1589565/Polish.v3.zip)

samdenty99, 2017-12-27:
No problem, fixed in a7a5531

the4anoni, 2017-12-27:
Thanks for patience :)

the4anoni, 2017-12-27:
Are U planning do 8.0 Polish version or in next version ?

samdenty99, 2017-12-27:
The translation engine was only added in `9.0` upwards, so to test it you'll need to flash it with Arduino or wait until I upload the compiled binary to the [releases page](https://github.com/samdenty99/Wi-PWN/releases/tag/9.0)

the4anoni, 2018-01-15:
U forgot to apply changes to demo. Also when cokpiled version will arrive ?

commit 26b2b97e (2017-12-27): :abcd: Polish translation

closes #42