samdenty/Wi-PWN #45: Not spamming beacons

Seldos, 2018-01-14:
It is not creating mass beacons just 2 or 3 after a minute of waiting. It worked with the spacehuhn version

edo-2313, 2018-01-16:
Just to be sure, did you enter some names in the list?
And are you using the beacon function or the random function?

StijnSimons, 2018-01-17:
Did you flash using Arduino or the .bin?

Seldos, 2018-01-20:
Used both functions added 48 times the same name (Clone). I used both cannot properly reset arduino flashed very often and used the arduino installation until it "worked" maybe that's the reason?

StijnSimons, 2018-01-24:
Try erasing your esp with ESPtool an reflash it.

Seldos, 2018-01-24:
worked thank you but why didn't it work when I used a 1mb file with 0xff and wrote it at 0x0 0x1 0x2 0x3?

StijnSimons, 2018-01-27:
Might have been a problem that the data got corrupted.

closed 2018-01-27 by StijnSimons