samdenty/Wi-PWN #47: Remote control using FTP

drdeft, 2018-01-16:
It could be perfect to contol device using remotely using config files by FTP.

Wi-PWN attacks list of networks for 10 minutes.

After it device connects to SSID with internet access by setted up password.

It loads TXT file on FTP server and writes there ID of itself and lists found target SSID:

===============================================
Wi-PWN Butovo 1 flat 13

Pasukova: skip
MkNet_24333: skip
HomeTel: skip
Vasya: skip
Veronika: not found for 3 days
RosTelecom_333f: skip

===============================================

You can change **skip** to **attack** to tell device to attack marked SSIDs and all clients connected to them.

After accessing to TXT on FTP server device continues to attack its targets.

edo-2313, 2018-01-16:
I think that it would be a completely different project, as it would have no interface (or very minimal) and it would require a server to control it

StijnSimons, 2018-01-17:
Takes up massive amounts of RAM too.