samdenty/Wi-PWN #49: Wifi Deauther OLED V3 Board is not Compatible

ChandraOrbit, 2018-01-20:
I did the testing 
[This file](https://github.com/samdenty99/Wi-PWN/releases/download/9.0/Wi-PWN_9.0_ENGLISH_OLED.bin) with [Wifi Deauther OLED V3 Board](https://www.aliexpress.com/item/WiFi-deauther-OLED-V2-include-Case-and-Antenna/32825138910.html?spm=2114.10010108.1000013.4.3fb2ad83sZo9kj&traffic_analysisId=recommend_2088_2_90158_iswistore&scm=1007.13339.90158.0&pvid=50500de9-2d1a-4141-9caf-d33ccea3c36d&tpp=1) but not Compatible
OLED does not appear any images and Wifi SSID does not appear

Do I have to Compile & install with Arduino manually and what should be changed in its source code ?

ChandraOrbit, 2018-01-21:
Solved .. 
Compile, change  
//create display(Adr, SDA-pin, SCL-pin)
 //SSD1306 display(0x3c, 5, 4); //GPIO 5 = D1, GPIO 4 = D2
  SH1106 display(0x3c, 5, 4); 

and Install using Flash Size 1 Mbyte

closed 2018-01-21 by ChandraOrbit

TimSSweeney, 2019-08-10:
wait can u help me?

TimSSweeney, 2019-08-10:
Is says those libraries can not be found

koko004, 2022-11-06:
> Is says those libraries can not be found

Use esptool to flash bin file and set this pinout

[image]