samdenty/Wi-PWN #50: Board OLED Error E19 and E95

ChandraOrbit, 2018-01-23:
I have done several test times on multiple boards:

1. For existing OLED boards, especially Wifi Deauther Oled V3 will experience errors during storage after making changes to the Settings menu, both E95 and E19 error errors.

2. For Non OLED board, error does not occur.

freesofia, 2018-02-19:
...the same problem.

loewal, 2018-02-22:
I also noticed, dat the ESP-01 (1 Mb. version) doesn't keep its settings. Several others do, like Wemos Lolin.
By the way:
Deauther 1.6 works perfect on my WeMos D1 ESP Wroom 02 ESP8266 NodeMcu Lua WiFi with Oled. Can you tell me why versions 8 and 9 with oled don't?

freesofia, 2018-02-23:
I use v3.0 board, but v8 and v9 with OLED not work correctly for me!
http://prntscr.com/iin80f

closed 2018-03-25 by samdenty99