samdenty/Wi-PWN #51: Hangs the browser after turning off the device

mrsergvit, 2018-01-27:
ESP8266
Android SDK API 19, 25
Google Chrome ver 63.0
Not tested on another browser.
Nothing a customized in the firmware.

For example:
Turn on the device, select SSID, enable testing of the attack with deauth.
After that, do not close the tab from the web page and the shutdown device (disconnect from a power).
It's done! You are lovely, Google Chrome will be frozen.

Eric00x, 2018-01-27:
Just don't use Google Chrome, you can download Wi-PWN app...

mrsergvit, 2018-01-27:
Well, I will test other browsers.
At the expense of the application - it is rather poorly developed due to the use of WebView.
In principle, I would be ready to write a good application, but now I haven't time to understand how this works. If there is a normal API documentation - please provide it and I will solve this issue at leisure and present it to the community.

Eric00x, 2018-01-27:
Look @up dude, Sam made the Wi-PWN app (it's on Play Store), so you don't need to code..

StijnSimons, 2018-01-27:
To be honest with you here, if you think its poorly developed, i would like to see how you improve it.
https://github.com/samdenty99/Wi-PWN/tree/master/android_app

mrsergvit, 2018-01-27:
Hhh, dudes.
First I tested the application on API 19, there was a problem with the text hint. He was at the top of the entered.
Now I have downloaded it to the API 25 - the application disappears or even the connection.
If you suddenly have free time, I'll try to debug it.

About sources, it is, of course, good, but the fact is that used WebView and already work, probably, from JS. In android app haven't a body for requests.

=====
Куфддн, I believe that it would be better to rewrite the application on a native level, without using WebView. But for this, I still need the API documentation of requests.

samdenty99, 2018-01-27:
If I was going to create a new app for it, I would use [React Native](https://facebook.github.io/react-native/) instead of Java as it's scalable and cross-platform (and I have loads of experience with React). 

The API documentation is non-existent, in order to develop Wi-PWN I had to inspect all the traffic myself to see how it handles requests. @spacehuhn *may* be releasing version 2.0 of the deauther -- which is being done from scratch, and he may end up using the Wi-PWN interface once he's got to that stage.

When / if he releases it and if he decides to use the Wi-PWN interface, I may decide to create a React Native application for it, but that's entirely up to him

closed 2018-03-25 by samdenty99