samdenty/Wi-PWN #56: Wifi works but No webpage on 192.168.4.1

aaa1niepodam, 2018-02-04:
ESP8266 cp2102 nodemcuv2

Hello like in title there's open wifi which i can connect but I cant open 192.168.4.1 webpage.
It is pinging back.
I did scan wifi devices and ports and It appears to be on 192.168.4.1 but no open ports (80 is missing)

StijnSimons, 2018-02-04:
Any logs?

aaa1niepodam, 2018-02-05:
Tell me where I can find it or how to prepare it. Im not as much technical/dev person.
I flashed NodeMcu and it showing Wifi AP ESP_xxxxxxx.
I am able to connect or ping but when trying to open webpage I am receiving connection refused (locked port or server offline?)

deltaxflux, 2018-02-11:
Please check out following commands. I hope you running Linux because otherwise, these will not work. However, I paste the windows alternative.

First, of all, we check if you got a valid IP.
`ifconfig | grep inet | grep 192`

On window, I think it was
`ipconfig`

Check if you got an IP address like 192.168.4.X. The default gateway should be 192.168.4.1.

If you don't get any IP here we go there is the problem. If another address appears, check if you got the captive portal side appears on this gateway address.

Next, we check if it responds.

In some cases, the web server does not respond. The DHCP is working fine will the web server has a bug. Check if you could ping 192.168.4.1

`ping 192.168.4.1`

Does it respond back or do you have any 100% package lost?

How to fix:
There are the ton of ways and it is not guaranteed that it works. However, here are some tips.

I recommend that you reflash your chip it is maybe because the transfer process was kind of invalid or something goes wrong. Please check your cable check that you have selected the right device and remove any other USB devices which maybe affect the process. 

If none work there could be something wrong with the chip itself. These aren't expensive so you could buy a check new one.

Hope it helps,

regards,

deltax

StijnSimons, 2018-02-11:
The bin flasher is always buggy. Always has been. Erase the memory using esptool.py and reflash using Arduino.

M4R3Kcracks, 2018-02-11:
Boi u didnt flash it thats the default ap that is ob the esp from factory

vuong244, 2018-02-25:
https://www.hackster.io/vuong244/deauthentication-attack-with-esp8266-4ed3d3

aaa1niepodam, 2018-03-13:
It works after few more flashes. Just needed to wait little more before it starts and create AP

closed 2018-03-13 by aaa1niepodam