samdenty/Wi-PWN #57: Network Does Not Show Up

EyesightTechnology, 2018-02-20:
I was trying to use the other popular NodeMCU deauther by spacehunn, when I encountered the frustrating and unsolvable problem that the network was not appearing.

So, I've tried this program and am sorry to say it has the same effect. No network is appearing in my network lists.

**My configuration:**
NodeMCU: https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

NodeMCU Flasher method

Baud: 230400
Flash Size: 512kb
Flash Speed: 40MHz
SPI Mode: DIO

This Program: V9.0

Currently, I have the module plugged into the USB of my laptop. **The BLUE LED is blinking about once per second, steadily. The RED LED is off.**

Any help would be greatly appreciated!!! Thanks!

loewal, 2018-02-20:
Hi,

Did you try to show its activity using the serial port?
I am using ESPlorer. Fine tool!
It shows exactly the properties with which you are monitoring.

EyesightTechnology, 2018-02-20:
Ok so I tried the tool. When I attempt to open the serial port the tool crashes. This is the output in the crash log:
`#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=14456, tid=11596
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.1+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: ESPlorer.jar

Host: Intel(R) Core(TM) i5-3427U CPU @ 1.80GHz, 4 cores, 3G,  Windows 10 , 64 bit Build 16299 (10.0.16299.15)
Time: Tue Feb 20 08:24:03 2018 Eastern Standard Time elapsed time: 7 seconds (0d 0h 0m 7s)

---------------  T H R E A D  ---------------

Current thread (0x0000017e4aa55800):  JavaThread "AWT-EventQueue-0" [_thread_in_native, id=11596, stack(0x00000042f8b00000,0x00000042f8c00000)]

Stack: [0x00000042f8b00000,0x00000042f8c00000],  sp=0x00000042f8bfabc0,  free space=1002k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]
C  [jSSC-2.8_x86_64.dll+0x12590]
C  [jSSC-2.8_x86_64.dll+0x1595]
C  0x0000017e350a919c

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  ESPlorer.ESPlorer.portOpen()Z+108
j  ESPlorer.ESPlorer.OpenActionPerformed(Ljava/awt/event/ActionEvent;)V+160
j  ESPlorer.ESPlorer.access$27300(LESPlorer/ESPlorer;Ljava/awt/event/ActionEvent;)V+2
j  ESPlorer.ESPlorer$271.actionPerformed(Ljava/awt/event/ActionEvent;)V+5
j  javax.swing.AbstractButton.fireActionPerformed(Ljava/awt/event/ActionEvent;)V+84 java.desktop@9.0.1
j  javax.swing.AbstractButton$Handler.actionPerformed(Ljava/awt/event/ActionEvent;)V+5 java.desktop@9.0.1
j  javax.swing.DefaultButtonModel.fireActionPerformed(Ljava/awt/event/ActionEvent;)V+34 java.desktop@9.0.1
j  javax.swing.JToggleButton$ToggleButtonModel.setPressed(Z)V+148 java.desktop@9.0.1
j  javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Ljava/awt/event/MouseEvent;)V+35 java.desktop@9.0.1
j  java.awt.AWTEventMulticaster.mouseReleased(Ljava/awt/event/MouseEvent;)V+8 java.desktop@9.0.1
j  java.awt.Component.processMouseEvent(Ljava/awt/event/MouseEvent;)V+64 java.desktop@9.0.1
j  javax.swing.JComponent.processMouseEvent(Ljava/awt/event/MouseEvent;)V+23 java.desktop@9.0.1
j  java.awt.Component.processEvent(Ljava/awt/AWTEvent;)V+81 java.desktop@9.0.1
J 4112 c1 java.awt.Container.processEvent(Ljava/awt/AWTEvent;)V java.desktop@9.0.1 (22 bytes) @ 0x0000017e35f1c6fc [0x0000017e35f1c3e0+0x000000000000031c]
j  java.awt.Component.dispatchEventImpl(Ljava/awt/AWTEvent;)V+496 java.desktop@9.0.1
J 4054 c1 java.awt.Container.dispatchEventImpl(Ljava/awt/AWTEvent;)V java.desktop@9.0.1 (129 bytes) @ 0x0000017e35eee51c [0x0000017e35eee3e0+0x000000000000013c]
J 4055 c1 java.awt.Component.dispatchEvent(Ljava/awt/AWTEvent;)V java.desktop@9.0.1 (6 bytes) @ 0x0000017e35eef4b4 [0x0000017e35eef3c0+0x00000000000000f4]
j  java.awt.LightweightDispatcher.retargetMouseEvent(Ljava/awt/Component;ILjava/awt/event/MouseEvent;)V+327 java.desktop@9.0.1
j  java.awt.LightweightDispatcher.processMouseEvent(Ljava/awt/event/MouseEvent;)Z+155 java.desktop@9.0.1
j  java.awt.LightweightDispatcher.dispatchEvent(Ljava/awt/AWTEvent;)Z+50 java.desktop@9.0.1
J 4054 c1 java.awt.Container.dispatchEventImpl(Ljava/awt/AWTEvent;)V java.desktop@9.0.1 (129 bytes) @ 0x0000017e35eee4bc [0x0000017e35eee3e0+0x00000000000000dc]
j  java.awt.Window.dispatchEventImpl(Ljava/awt/AWTEvent;)V+19 java.desktop@9.0.1
J 4055 c1 java.awt.Component.dispatchEvent(Ljava/awt/AWTEvent;)V java.desktop@9.0.1 (6 bytes) @ 0x0000017e35eef4b4 [0x0000017e35eef3c0+0x00000000000000f4]
j  java.awt.EventQueue.dispatchEventImpl(Ljava/awt/AWTEvent;Ljava/lang/Object;)V+41 java.desktop@9.0.1
j  java.awt.EventQueue.access$500(Ljava/awt/EventQueue;Ljava/awt/AWTEvent;Ljava/lang/Object;)V+3 java.desktop@9.0.1
j  java.awt.EventQueue$3.run()Ljava/lang/Void;+32 java.desktop@9.0.1
j  java.awt.EventQueue$3.run()Ljava/lang/Object;+1 java.desktop@9.0.1
v  ~StubRoutines::call_stub
J 1706  java.security.AccessController.doPrivileged(Ljava/security/PrivilegedAction;Ljava/security/AccessControlContext;)Ljava/lang/Object; java.base@9.0.1 (0 bytes) @ 0x0000017e3cbea646 [0x0000017e3cbea5e0+0x0000000000000066]
J 3863 c1 java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Ljava/security/PrivilegedAction;Ljava/security/AccessControlContext;Ljava/security/AccessControlContext;)Ljava/lang/Object; java.base@9.0.1 (22 bytes) @ 0x0000017e35e6ae8c [0x0000017e35e6aa20+0x000000000000046c]
[... 447 lines cut ...]
Launcher Type: SUN_STANDARD

Logging:
Log output configuration:
#0: stdout all=warning uptime,level,tags
#1: stderr all=off uptime,level,tags

Environment Variables:
[... cut ...]
OS: Windows 10 , 64 bit Build 16299 (10.0.16299.15)

CPU:total 4 (initial active 4) (2 cores per cpu, 2 threads per core) family 6 model 58 stepping 9, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, aes, clmul, erms, ht, tsc, tscinvbit, tscinv

Memory: 4k page, physical 3883924k(832972k free), swap 7160724k(2334624k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11) for windows-amd64 JRE (9.0.1+11), built on Sep 28 2017 04:54:03 by "mach5one" with MS VC++ 12.0 (VS2013)

END.
`

loewal, 2018-02-20:
> Event: 7.746 Thread 0x0000017e499ae000 4123 3 java.util.Formatter$FormatSpecifier:: (259 bytes)

Looks like a powerproblem. 
What is your source?

Verstuurd vanaf mijn iPhone 7 Plus

EyesightTechnology, 2018-02-21:
Well I was using it on my computer, that's how I got any serial communication in the first place!
So... I guess that means it was powered by Laptop USB

vuong244, 2018-02-25:
https://www.hackster.io/vuong244/deauthentication-attack-with-esp8266-4ed3d3

digmorepaka, 2018-02-25:
Try using esptool to erase flash and flash the binary with the following settings: Baudrate 115200
Flash Size: 4mb
Flash Speed: 80MHz
SPI Mode: DIO

dudenugget2000, 2018-02-26:
I have this same exact issue and board. Both repo's have not worked for me as well...

closed 2018-03-25 by samdenty99