samdenty/Wi-PWN #58: Connection Resetting

sayak-brm, 2018-02-22:
The connection is resetting for several files like `/js/functions.js` and `/main.css`, leaving the page partially rendered as shown below.
[image]

loewal, 2018-02-22:
I flashed the bin, because the source doesn't compile without errors. Dsniffer....

sayak-brm, 2018-02-22:
@loewal Wrong issue. I'm facing this issue on a 4MB NodeMCU.

For those interested, here's the network stats:
[image]

vuong244, 2018-02-25:
https://www.hackster.io/vuong244/deauthentication-attack-with-esp8266-4ed3d3

closed 2018-03-19 by sayak-brm