samdenty/Wi-PWN #69: Random TX Power for beacon attack

Enziak, 2018-06-14:
Hi.

It's possible to add random tx power for beacon attack like in spacehuhn project?
Thats the only code i find for it

` if (settings.getRandomTX() && (beacon.active || probe.active)) setOutputPower(random(21));
    else setOutputPower(20.5f);`

Could someone help with code and where to add it for arduino file?
Just add random tx power by default for beacon attack

StijnSimons, 2018-06-14:
https://github.com/samdenty99/Wi-PWN/blob/master/arduino/Wi-PWN/Attack.cpp

Enziak, 2018-06-14:
I'm still learning and can't

Enziak, 2018-06-14:
I'm still learning and I can't manage to do that.

`C:\Users\enzo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/WString.h:113: undefined reference to `setOutputPower(float)'`

I hope it's good idea to implement and will be in next release.

StijnSimons, 2018-06-16:
I'm not completely sure but i think the GPIO pins D0 and D1 are actually the built-in leds too.

Enziak, 2018-06-16:
Yeah, when connected to some stronger 1A+ usb, second led above rst button can be turn on or off when i touch GPIO pins D0 and D1 (really low constant light)

No output in serial monitor and can't notice if it do something more.

StijnSimons, 2018-06-16:
Feel free to make a pull request. Until then, i’m closing this issue as it is no longer necessary right now.

closed 2018-06-16 by StijnSimons