samdenty/Wi-PWN #71: Deauth detector not working

mijanto, 2018-07-21:
When I start "Detector" it just restarts ESP and nothing happens. My chip is CP2102 and it's doing the same on CH340G.

samwedge, 2018-08-12:
I have a similar experience with CH340G. Not sure if it's restarting or not, but the LED does not light up when I deliberately launch a death attack from another device. The deauth attack is confirmed working as it knocks my PC off the network.

samwedge, 2018-08-12:
OK, I can now see it resets. Rather than monitoring for DeAuth frames, I can see that the ESP has reset and is acting as an AP again.

Oddly, by repeating the process of entering "Deauth Detector" mode over and over again, it eventually works on a one-off basis. It looks as if it is drawing too much instantaneous power when entering Monitor Mode, causing the voltage to drop and the ESP to reset

gojigeje, 2018-08-23:
I also experiencing same problem on my ESP-01 board, it resets when starting detector.

I got it working by placing small delay [on this line](https://github.com/samdenty99/Wi-PWN/blob/master/arduino/Wi-PWN/Wi-PWN.ino#L506) (100ms)

I don't know why it is working though.

victordetarragon, 2019-02-26:
Same issue. Is the project dead ?

StijnSimons, 2019-03-29:
Kinda is, but i'll look into it now. Who knows it might come back to life? 😉

StijnSimons, 2019-03-31:
Added workaround for now.