samdenty/Wi-PWN #79: getting irom0.text' will not fit in region `irom0_0_seg' error when compiling

codehackr, 2018-10-04:
Anyone know how to fix this?

StijnSimons, 2018-10-05:
Maybe put the full log?

codehackr, 2018-10-05:
how do i get the full log?

StijnSimons, 2018-10-05:
Just copy and paste the whole output.

codehackr, 2018-10-05:
Arduino: 1.8.5 (Windows 7), TD: 1.40, Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (1M SPIFFS)"

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\LATITU~1\AppData\Local\Temp\arduino_cache_40453\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M1M_31fa24bae33cb11efadcef56b3248f92.a
c:/users/latitude e6430/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\LATITU~1\AppData\Local\Temp\arduino_build_231192/Wi-PWN.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

codehackr, 2018-10-05:
Ive made all the changes in the howto and reverted to SDK 2.0.0

codehackr, 2018-10-07:
any ideas?

codehackr, 2018-10-07:
Arduino: 1.8.5 (Windows 7), TD: 1.40, Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (1M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Latitude E6430\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Latitude E6430\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Latitude E6430\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M1M -ide-version=10805 -build-path C:\Users\LATITU~1\AppData\Local\Temp\arduino_build_332453 -warnings=none -build-cache C:\Users\LATITU~1\AppData\Local\Temp\arduino_cache_147508 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Latitude E6430\Downloads\WiFiDeauther\Wi-PWN\Wi-PWN.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Latitude E6430\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Latitude E6430\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Latitude E6430\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M1M -ide-version=10805 -build-path C:\Users\LATITU~1\AppData\Local\Temp\arduino_build_332453 -warnings=none -build-cache C:\Users\LATITU~1\AppData\Local\Temp\arduino_cache_147508 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Latitude E6430\Downloads\WiFiDeauther\Wi-PWN\Wi-PWN.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Using core 'esp8266' from platform in folder: C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Detecting libraries used...
"C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "C:\Users\LATITU~1\AppData\Local\Temp\arduino_build_332453\sketch\Wi-PWN.ino.cpp" -o "nul"
"C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "C:\Users\LATITU~1\AppData\Local\Temp\arduino_build_332453\sketch\Wi-PWN.ino.cpp" -o "nul"
"C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardw [... cut ...]
Using library ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display at version 4.0.0 in folder: C:\Users\Latitude E6430\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display 
Using library EEPROM at version 1.0 in folder: C:\Users\Latitude E6430\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\EEPROM 
Using library deauth in folder: C:\Users\Latitude E6430\Documents\Arduino\libraries\deauth (legacy)
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

StijnSimons, 2019-01-14:
You either don't have enough space on your device or your chip is fried.

StijnSimons, 2019-01-14:
For more information check the issue [here](https://github.com/spacehuhn/esp8266_deauther/issues/879).

closed 2019-01-14 by StijnSimons