samdenty/Wi-PWN #86: Add captive portal

aniketstark, 2019-02-16:
Add captive portal in wifi pwn so its look like cool like wifi phisher or fluxion so please pull the new update

samdenty, 2019-02-16:
It has it. You need to enable a flag in wi-pwn.ino

closed 2019-02-16 by samdenty

aniketstark, 2019-02-17:
H. I dont understand bro

minanagehsalalma, 2019-02-17:
@samdenty  should I Remove the # before every captive portal word in the code ?!

StijnSimons, 2019-02-17:
https://github.com/samdenty/Wi-PWN/blob/ab43f87b521bbac2507a1bfd6b8fa25389599767/arduino/Wi-PWN/Wi-PWN.ino#L41

aniketstark, 2019-02-17:
Is that work

minanagehsalalma, 2019-02-17:
Test it by yourself dude .. Remove the #

aniketstark, 2019-02-17:
What @minanagehsalalma

aniketstark, 2019-02-17:
[image]

aniketstark, 2019-02-17:
What errase

aniketstark, 2019-02-17:
Now i am testing wait i will back

aniketstark, 2019-02-17:
Nothing not showing captive portal option

minanagehsalalma, 2019-02-17:
Dude. .did you remove the # ? From the 41 line ?
. it should show you the captive portal when you visit any website !!

aniketstark, 2019-02-17:
But aaa  i dont understand i remove it like this 
#define USE_CAPTIVE_PORTAL  /* <-- enable captive portal (redirects all pages to 192.168.4.1) - most devices flood the ESP8266 with requests */
Or
Like this 
define USE_CAPTIVE_PORTAL  /* <-- enable captive portal (redirects all pages to 192.168.4.1) - most devices flood the ESP8266 with requests */
Which one ?

And i edit like 1st one so where is captive portal option ????

minanagehsalalma, 2019-02-17:
//define USE_CAPTIVE_PORTAL

minanagehsalalma, 2019-02-17:
> And i edit like 1st one so where is captive portal option ????
there isn't 

> enable captive portal (redirects all pages to 192.168.4.1) 
it will just redirects  the web pages requests to  192.168.4.1 automatically...
you can test it by yourself by connecting to the wifi then visit any website and see if it redirects you to 192.168.4.1  or not !!!!!!!!!!!

aniketstark, 2019-02-17:
So thats all its not work like fluxion
You say-:you can test it by yourself by connecting to the wifi ("it means i need to enable a wifi client mode ah ha") to redirect to another one

aniketstark, 2019-02-17:
Therw is no option like captive portal

minanagehsalalma, 2019-02-17:
Dude what are you saying !!!
Just use another phone to test it ..
BTW you look like you understand nothing !!! Have you read the fluxion
requirements before ?!

aniketstark, 2019-02-17:
Oh sh^_^t i use another phone to test it

minanagehsalalma, 2019-02-17:
"captive portal
To create fake ap and redirect to page and get creditional" 
You sound stupid !! 
You just need a captive portal then you can design the page by yourself .. Or copy it from fluxion .. ..

aniketstark, 2019-02-17:
But not work which option 
I see only these option wait is sending you

aniketstark, 2019-02-17:
Okok  wait wait  i am stupid ok i am noob🤓
Thats all
But yarr(bro) if you know all these thing so send me na video create video and send me your pro😎😎

minanagehsalalma, 2019-02-17:
I have already sent you a video !!

minanagehsalalma, 2019-02-17:
Just flash this one https://github.com/exploitagency/github-ESPortal

aniketstark, 2019-02-17:
Bro bro you don't understanding that's why i don't understanding i need captive portal in WiFi pwn tool to use

aniketstark, 2019-02-17:
I don't want to flash new tool again again and again

minanagehsalalma, 2019-02-17:
You should have 2 esps ...
Because it would be too much. For one esp to do !!

aniketstark, 2019-02-17:
Why dont with one esp8266

minanagehsalalma, 2019-02-17:
Because it will be less complicated ... And easier for the esp to handle !!
And because there is no code that is available outside to do both at the
same time !!! Maybe you should learn arduino programming and write your own
tool !!

StijnSimons, 2019-02-17:
You don’t NEED a second one. Just uncomment the def in the arduino file and compile the project.

StijnSimons, 2019-02-17:
In case this is too hard. Your line should look like this
`#define USE_CAPTIVE_PORTAL  /* <-- enable captive portal (redirects all pages to 192.168.4.1) - most devices flood the ESP8266 with requests */ `

minanagehsalalma, 2019-02-18:
@StijnSimons  ..... Can the the captive portal page be edited to receive plaintext ?!
Where is the path of it's file ?!

samdenty, 2019-02-18:
This issue is off-topic for the scope of this project.