samdenty/Wi-PWN #89: resetting the Deauther wifi ID and password

forestkunt, 2019-03-09:
A friend of mine purchased one of your deauther chips and in true noob fashion changed the settings, wfi id and password then totally forgot all of them. Can you help me out with a root password or something so i can get him off my back to fix this for him. Here is a photo of the chip
[image]

cheers

StijnSimons, 2019-03-10:
These chips aren't ours. Either way, if you connect it to your PC using a microUSB cable, you can see the password and ssid in the serial console.

closed 2019-03-11 by StijnSimons

eastcoastboys, 2019-09-20:
@StijnSimons How do we use the serial console to look at the SSID and password