samdenty/Wi-PWN #90: JS/HTML injection via SSID

jn4kr, 2019-03-17:
As you can see it's possible to inject HTML/Javascript in the attack page via the ssid. 

I don't consider it as a future.

### Screenshots
[image]

[image]

### Steps to reproduce

1. Create a beacon with the tag you wan't to inject as SSID (e.g. <svg onload=\"alert('1')\"/> )
2. Select the beacon as target
3. Switch to the attack page

minanagehsalalma, 2019-03-20:
@jn4kr  ...
Does android even process these WiFi names in the available wifis list ?!

jn4kr, 2019-03-20:
Yeah. But i think it's off topic, so I'll sent you a mail about this topic.

StijnSimons, 2019-03-21:
Project is kinda dead but once i really don’t know what to do, i’ll think about it.

kingdevnl, 2019-04-05:
We might be able to strip out all HTML code using a REGEX?

Helixable, 2019-10-31:
Check thsi out: https://code-examples.net/en/q/129e949