samdenty/Wi-PWN #93: 0pkts/s on Arduino code and 20pks/s on .bin image

naamah75, 2019-06-03:
First flashed the no LCD .bin image, the deauth attack shown 20pkts/s, then I tried to use the Arduino source; flashed it, no apparent problem, the web interface is responsive, the wifi scan get the same results as the .bin image but the attack remain freezed to 0pkts/s. 
The output in the serial terminal show no error, it seems all working good.

StijnSimons, 2019-06-03:
What board are you using? NodeMCU 0.9?

naamah75, 2019-06-03:
Wemos D1 mini Pro 1.0.0

kingdevnl, 2019-06-07:
Iv'e never seen that board before, 
Are you sure it has an ESP8266 module?

StijnSimons, 2019-06-07:
Might be something hardware related. For now the project is only targeted to the NodeMCU 0.9-1.0 series.

naamah75, 2019-06-07:
I'm sure that the board is based on ESP8266 https://wiki.wemos.cc/products:retired:d1_mini_pro_v1.0.0.
The strange fact is that the precompiled .bin file works like a charm, this demonstrate that the board is compatbile with this project, and, in addition, it work also partially when try to flash with the arduino sources; in this case all seems to work except that the deauth attack.
Everything lets imagine that, maybe, the sources code for ardino isn't aligned with the newest .bin image.

StijnSimons, 2019-06-07:
The problem is that you’re probably compiling for the NodeMCU board.

```
Select your ESP8266 board at Tools > Board and the right port at Tools > Port If no port shows up you need to reinstall the drivers, search online for chip part number + 'driver Windows'
```

naamah75, 2019-06-07:
Nope! Is selected the right board... I used a lot times the WeMos board and I know very well how to use.

yentzee, 2019-08-28:
Same problem over here :(