samdenty/Wi-PWN #116: wi pwn

ales197311, 2021-11-23:
прошу прощения. У меня стоит esp 8266 версия 3.0.2. Как установить версию 2.0.0?
И что означают deauth, beacon, random, probe-request, как ими пользоваться? Где почитать?

osysltd, 2021-11-23:
Дружище, не задавай дурацких вопросов пожалуйста, Тикеты в гитхабе созданы для другого. кури мануалы, написано же всё https://github.com/samdenty/Wi-PWN/blob/master/README.md

Dear friend, please do not ask stupid question, tickets in GitHub are for different purposes, Read the f*cking manual, RTFM.

closed 2021-11-23 by osysltd

ales197311, 2021-11-24:
Ну извини, я начинающий просто, если честно, даже до конца не понял, для чего ардуинки нужны, просто захотел сделать глушилку для вай фая )))