samdenty/Wi-PWN #124: Error while verifying in ArduinoIDE

HECKER723626, 2023-03-02:
First of all i would like to state that i am a noob at all of this and some of my words might not make sense so bear with me a bit.
The code that had to be uploaded to the nodemcu was stopped due to an error. 

C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\APScan.cpp: In member function 'bool APScan::start()':
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\APScan.cpp:18:59: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'
   results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)
                                                           ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\APScan.cpp:18:59: note: candidate is:
In file included from C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\APScan.h:6:0,
                 from C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\APScan.cpp:1:
C:\Users\asm24\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)
     int8_t scanNetworks(bool async = false);
            ^
C:\Users\asm24\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\Attack.cpp: In member function 'bool Attack::send()':
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\Attack.cpp:109:50: error: 'wifi_send_pkt_freedom' was not declared in this scope
   if (wifi_send_pkt_freedom(packet, packetSize, 0) == -1) {
                                                  ^

exit status 1

Compilation error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

After this error i tried to get this code run on version 3.1.1 instead of 2.0.0. after that there was another error in the last quarter of the code.

C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp: In member function 'String NameList::get(int)':
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:63:23: error: qualified-id in declaration before '(' token
   63 |   String NameList::get(int num) {
      |                       ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:72:21: error: qualified-id in declaration before '(' token
   72 | Mac NameList::getMac(int num) {
      |                     ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:76:26: error: qualified-id in declaration before '(' token
   76 | String NameList::getByMac(Mac client) {
      |                          ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:87:26: error: qualified-id in declaration before '(' token
   87 | int NameList::getNumByMac(Mac client) {
      |                          ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:91:22: error: qualified-id in declaration before '(' token
   91 | void NameList::remove(int num) {
      |                      ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:102:20: error: qualified-id in declaration before '(' token
  102 | void NameList::edit(int num, String name) {
      |                    ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:107:1: error: expected '}' at end of input
  107 | }
      | ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:53:31: note: to match this '{'
   53 | String NameList::get(int num) {
      |                               ^
C:\Users\asm24\OneDrive\Documents\Arduino\Wi-PWN-master\arduino\Wi-PWN\NameList.cpp:107:1: error: control reaches end of non-void function [-Werror=return-type]
  107 | }
      | ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: qualified-id in declaration before '(' token

Now to get proper understanding i tried chatgpt and corrected some codes there but still no luck.Another thing to mention is that i was not able to find Method 2 step 9(more prreferences) like it was specified

andre2l2, 2023-07-25:
I getting some error

Renan0eng, 2024-01-26:
I had this same problem with Arduino IDE 2.2.1. The fix was simple: just move the return statement of the function 'NameList::get(int num)' from line 60 to 61 in the file 'NameList.cpp'. 
Here's the code

```
String NameList::get(int num) {
  String returnStr;
  if (num >= 0) {
    for (int h = 0; h < nameLength; h++) {
      if (names[num][h] != 0x00) returnStr += (char)names[num][h];
    }
    returnStr.trim();
  }
  return returnStr;
}
```

I hope it's the same error as mine, good luck, I hope I've helped.